wezhunter/ESP32_LinuxCNC_MotionController_RealTime #3: spindle encoder feedback for lathe and rigid tapping

jschoch, 2024-02-13:
Encoder should update counts ( int64 would be nice) as well as send velocity units so that G96 can work.

remora has encoders as does linuxcnc-rio for reference.

PCNT could be used but it is limited to int32 AFAIK.  It is polling, and wouldn't work for my ELS but i don't think linuxcnc needs super fast updates for velocity and maybe not for position (G33)

jschoch, 2024-02-26:
very cool!

how is the latency for the udp over usb?  I didn't think usb was viable for realtime....

Also, this is great for closed loop motion, but spindle encoder also typically needs an index pulse and it needs to have a LCNC component to help calculate velocity.  You aren't trying to control the spindle position so much as map the Z axis to the spindle movement.  There doesn't need to be a closed loop velocity between the spindle and the spindle motor.

wezhunter, 2024-02-26:
Thanks for the info. Yeah I'm aware of the required indexing :) That will be included since it involves using the same underlying framework. Ideally including lathe motor/mill spindle RPM is best, for those that have encoders, including index, to ensure it's operating within spec. Since it's possible to include it - then why not?

wezhunter, 2024-02-27:
Regarding the latency over USB please could you ask the question over in the open UDP issue so others who visit this project can see it? Thanks :)

jschoch, 2024-02-28:
maybe just this links them

https://github.com/wezhunter/ESP32_LinuxCNC_MotionController_RealTime/issues/14