wezhunter/ESP32_LinuxCNC_MotionController_RealTime #14: Implement ESP32-S2/S3 USB device mode RNDIS/ECM virtual network adapter support

wezhunter, 2024-02-18:
ESP32-S2/S3 does not have an Ethernet PHY IP block in silicon.
Instead it has a USB PHY.
USB RNDIS/ECM virtual networking adapter support has been proven to work well with low latency and reliably.

As this project uses UDP to communicate to the host then this would be an ideal feature which allows for a true plug-and-play controller.
Reliable USB CDC Serial console needs to be implemented too (USB multi-endpoint).

FastAccelStepper library supports RMT on the S2 & S3 MCUs.
S2 supports up-to 4 steppers via RMT. S3 supports up-to 8 steppers via RMT.
S2 is single core
S3 is dual core (like ESP32)

Ideally S3 USB networking support provides the ultimate solution. Dual core, plenty of IO (44 vs 34), ESP-NOW, low cost and abundantly available.

USB device networking support frees up a decent amount of pins on the S2/S3 for stepper motors or inputs that would otherwise be used by either SPI or RMII ethernet PHY.

wezhunter, 2024-02-21:
Initial build with USB Device Mode both ECM or RNDIS (virtual networking) support for ESP32-S2 built on Arduino framework has been proven to work with excellent latency. Development required a bespoke USB network driver within esp-idf to ensure LWIP and esp_netif stack is handled properly.

Hardware based step gen via RMT on 4 steppers (max RMT channels for S2) are all generating consistent simultaneous clean pulses with similar results to native RMII phy! Console via UART0 is reliable and consistent whilst step gen is running continuously. All on a single core (S2 is single core only)!

Additional tweaks and cleanup required to ensure all potential error conditions and MCU hard and soft restarts are handled properly (USB phy registers etc) so that any restart of the board ensures networking stack inits every time.

USB CDC ACM serial port console (USB COM port for Console CLI) is yet to be tested; it may saturate the CDC/NET endpoint xfers whilst motion is running on a single core and so may require designing around for the S2 specifically but there are plenty of ways to solve (rate limiting, Telnet only etc)

A seamless, USB connected, LinuxCNC controller is officially confirmed as being possible

trihwangyudi, 2024-02-23:
so its will use usb type C for comunicating with machine (or board), will wait for real.....
maybe its the simple and most cheaper solution, have to say greats project, 😁😁

wezhunter, 2024-02-23:
Thanks :)
Yes that's correct! No ethernet. Just one cable. Controller is powered by the host too. 

USB 5V ground loop isolators will work if needed, however generally speaking EMI in noisy environments can be mostly solved with a screened & shielded USB cable.

I've ran a 24 hour CNC job on a full 6 axis ESP32-S2 (single core) bench setup with all 6 motors connected.
ESP32-S2 is connected to a Raspberry PI 4 running LinuxCNC purely via USB (virtual Ethernet). 

Each of the motors speeds are set to 60khz pulse rate per axis (to stress test it) and none have overflowed nor underran so it's not missed a single step! Latency is excellent and jitter is surprisingly low - on par with native Ethernet RMII so pretty much as good as you can get.

It's been a lot of work to get working and still quite a bit to do to get it right

wezhunter, 2024-02-23:
Videos of each test will be uploaded sometime tomorrow evening...

trihwangyudi, 2024-02-23:
greats ... even its just 4 axis its still awesome project for me (actually 4 axis is more than enough for me).... i have start to build a new machine... so its will be ran linuxcnc, with yours esp project, thanks for yours greats jobs... maybe i will make a video and have tag for yours project

wezhunter, 2024-02-26:
USB Axis f-error rates (no jitter during moves)
https://youtu.be/nzYs2vdiTkY

Wireless XHC MPG Pendant USB and USB S2 motion controller on same root hub latency 
https://www.youtube.com/watch?v=rgdr76XOblc

wezhunter, 2024-02-27:
The current usb_net driver solely relies on ESP-IDF APIs - nothing Arduino.

In order to hook onto the Arduino "Ethernet" event handlers for IP Address & Link State handlers to easily determine if USB Ethernet is running etc I've had to convert the usb_net driver over to C++ which was quite a bit of work, determining correct atomic methods was fun.

This refactor now permits USB Networking to be brought up without any WiFi initialisation dependencies whatsoever - so use-cases that don't want to have WiFi running at all doesn't have to. 

It's now able to call Arduino C++ `tcpipInit()` directly from the usb_net driver - just like existing Ethernet drivers.

Arduino stack continues to work as normal the USB networking is now just another native Ethernet adapter internally.

rohit1974, 2024-04-11:
Sir, 
driver , udp.comp not  install I try many time but always found error user/bin/halcomplie not found udp.comp file