wezhunter/ESP32_LinuxCNC_MotionController_RealTime #15: Motion: Improve initial STEP pulse train jitter on motor ramp up

wezhunter, 2024-02-18:
Motor positioning with FastAccelStepper library is perfect. 
However acceleration on an initial ramp up of a motor results in a small amount of pulse train jitter on the step pin until the motor reaches its target speed (begins to coast). 
The result is that the motor accelerates and decelerates between10-20ms whilst it ramps up and can be heard if a motor is clamped to a desk on high speed moves.

A combination of UDP latency and position command size windows causes the acceleration to start to clip in FAS since the velocity vs position vector window is essentially overflowing.

During development this was found to be due to the way LinuxCNC issues commands compared to what FastAccelStepper library requires whilst using its built-in positioning mode ``moveTo()``. Alternate positioning or movement commands in FAS do not suit the LinuxCNC use-case and the `moveTo()` feature works so well a solution ideally wants to be found.

LinuxCNC generates a velocity command that ramps up from 0 to the destination max velocity during a move. FAS expects the maximum vel value for a move from the start in order for it to perform smoothing of the step pulse train until coasting.

Confirmed that using LinuxCNC "vel_limit" HAL parameter, which is the current max feedrate, as the input speed to FAS and pulse train is clean throughout.

Unfortunately vel_limit does not update during a job with any feedrate overrides, for it's own reasons.

Investigate a solution so that `cmd.vel` is always the dynamic axis maximum speed value including any overrides in LinuxCNC.

jschoch, 2024-02-26:
I ported some of https://github.com/luni64/TeensyStep to use RTM here: https://github.com/jschoch/ESPels/blob/dev/src/src/Stepper.h

it uses two timers, one to deal with acceleration and another for calling RTM.

Maybe worth you looking at?  I had issues trying to use FAS for my els project.

wezhunter, 2024-02-26:
Thanks for the feedback. FAS works fine. As I've explained I already know what is up but just had to figure out how to ensure in "maps" to LinuxCNC. The jitter is now fixed in the next release once I'm happy with the release.

FAS is being used for precise positioning since it does it so well. I've torture tested positioning on a real machine using DTIs across all axes. There's zero step loss which is the only reason why I'd really consider using it in the first place.

Thanks

wezhunter, 2024-02-26:
Out of curiosity - what kind of problems have you experienced with the FAS library? Useful to know your own experiences

wezhunter, 2024-03-02:
Video proof of S2 running smooth motion at ridiculous speeds.
Units in LinuxCNC are real world representation.
Driver running at 1600ppr, scaled at 320 for a 5mm ballscrew.

All 4 axis on S2 can run at these speeds up to a combined 200khz.

https://youtu.be/KoUIp38FQs0?si=xKBBZnYa-j27Pxhl